owncloud-archive/pi-image #3: Use Let's Encrypt to set up SSL via a script

enoch85, 2015-12-30:
I have made a script that do 4 tries to setup SSL in Ubuntu. I released it on my [testing VM with PHP 7](https://www.techandme.se/pre-configured-owncloud-installaton/). I wrote about it in [this post](https://www.techandme.se/owncloud-8-2-1-php-7-redis-cache-lets-encrypt-webmin/).

I figured that we maybe could use this to install SSL on the rPI as well?

The script is just a first writeup, and I have not get it to do the setup properly yet. So it would be very cool of we could collaborate on this one, and use Let´s Encrypt to get a proper SSL cert on the Pi device.

You can find the script [here](https://github.com/enoch85/ownCloud-VM/tree/master/lets-encrypt).

Lets start hacking ;)

enoch85, 2016-04-04:
Update: The script is now updated and confirmed working.

ezraholm50, 2016-04-04:
Do note that I had to forward port 443 to the Pi for it to work.

oparoz, 2016-05-05:
Please re-open on https://github.com/kyrofa/owncloud-snap/issues

closed 2016-05-05 by oparoz