owncloud-archive/pi-image #4: Choice of the webserver

mmattel, 2015-12-31:
The PI has limited resources and to deal with this resources efficiently, I thougt it is worth to think about which webserver will be chosen.

When I used Apache, I found many things beeing reatively complex to handle and the stack itself is heavy. In a positive way, Apache can dynamically include functionalities. Some years ago I switched over to nginx and found that it is leightwight and easy to configure. And there are some features I do not like to miss when serving owncloud which is eg cache-purge. The speedup on presenting pictures was imressive. In nginx, you have to compile your functionalities in, which is what I have identified a easy process and described in the part of our webserver tips (cache-purge), but a task to be done.

Please see a performance comparison for those two wenservers:
http://raspberrywebserver.com/raspberrypicluster/comparing-the-performance-of-nginx-and-apache-web-servers.html

enoch85, 2016-01-01:
Are the comparisons made with PHP 7?

mmattel, 2016-01-01:
At the PHPCON in Tokyo October 2015, they also used nginx for their comparisons with the various PHP versions and HHVM. Pls see: http://talks.php.net/tokyo15#/boxspecs
I think that both PHP7 and nginx are very good for systems with limited ressources to get the optimum out of it.

oparoz, 2016-01-05:
The problem with some of these benchmarks is that they don't give any details on how the web servers were configured, so if it's Apache + mod_php vs NGINX+PHP-FPM, it's not a fair comparison.
Also, you're never going to have 25 concurrent connections on those things unless you're willing to disappoint all users, so I don't think it makes much of a difference at that stage. Obviously, the real test will be with a real image and some proper acceptance tests.

If NGINX is found to be beating Apache by a wide margin and if the target for the drive has no tech knowledge and software is completely controlled by a 3rd party, then you can go with NGINX as toying with the OS would be like breaking the warranty. On the other hand, if you expect people to tune their OS, then the only officially supported web server is Apache and that won't change unless some experts volunteer their time to configure and tune NGINX for owncloud. Some doc here and there doesn't count, it needs to be used in production, with different apps and shown to work flawlessly and to be as secure as the Apache solution.

oparoz, 2016-03-21:
The 1st image uses Apache 2.4.

closed 2016-03-21 by oparoz