owncloud-archive/pi-image #5: Make it possible to access cloud from anywhere

miska, 2015-12-31:
People might want to access the cloud not only from home and they might not have public IPv4. So I'm working on a way how to get them IPv6 and dyndns. You can see the progress here:
https://github.com/miska/ocipv6
And more details are provided here:
http://michal.hrusecky.net/2015/12/getting-ipv6-for-your-owncloud/

miska, 2016-01-01:
Dynv6 integrated, so there is a way to easily get permanent domain name.

miska, 2016-01-16:
Added support for uPnP to my app, testing and visual appearance fixes (as other fixes) are welcome:
https://github.com/miska/ocipv6/commit/90b4e9eae2203c214394ba1c6b72201d5d89ed78

enekux, 2016-01-24:
Awesome job @miska!
Before reading this post I also started to do my own research regarding UPnP and run into same client "miniupnpc" that you are using in your "OC IPv6" app. But of course you are far ahead doing the owncloud app... great!
From my tests I have noticed that if the router is rebooted the rules created by "miniupnpc" are lost... and I think that is not ideal. Instead, if the rules are manually added they stay.
I haven't found any special command in "miniupnpc" to make the rules permanent, the only way is or to add bash script to be run when interfaces are up or better to run cron job.
Here you have some examples:
https://pavelfatin.com/access-your-raspberry-pi-from-anywhere/

oparoz, 2016-05-05:
@miska Is the todo list up to date?