owncloud-archive/pi-image #6: Where to install the OS?

enoch85, 2016-01-01:
There have been some discussions in the mailing list about where to install the OS, and that the Pi eats SD cards and so on. 

I just got my Pi up and running, and during first time setup it asks where I want to install the OS. Since I never used a Pi before, I don't know how it works. Is it possible to install everything on the HDD, or is it a better solution to boot from the SD card, and in that case - how do I set it up to do that?

I will use [Ubuntu Snappy](https://developer.ubuntu.com/en/snappy/start/raspberry-pi-2/) as OS on the Pi, and right now I have issues with how to get Snappy on the Pi, and where to install it. Some advice would be great!

enoch85, 2016-01-02:
After a bit of digging on the internet, I think this would need to be scripted - so that when you as the end-user receive the Pi you just plugin the SD-card with the image on, and the HDD, boot it and the script starts.

I also noticed that there are different options:
- Run solely on the HDD
- Boot with the SD card and have the rest on the HDD (Have not figured this out yet, help wanted)
- Run everything on the SD card (not an option)

So basically, the script would come pre-installed on the SD card and it would handle everything for the user. That is:
- Make the HDD the main disk, and move the root partition and everything on the SD card to the HDD. [Here is a good tutorial for that](https://www.raspberrypi.org/forums/viewtopic.php?f=26&t=10914&p=129474&hilit=resizefs#p122476).
- Fetch the pre-made image from Github via wget, and then install it on the HDD. 
- If the image is built from scripts, then get all the scripts by putting a lot of if arguments in rc.local so that on first boot it would get the latest scripts from Github, run them and install everything needed. Like [I do today](https://github.com/enoch85/ownCloud-VM/tree/master/production) with the ownCloud VM.

Any other options, or suggestions to do this?

miska, 2016-01-02:
I think solely HDD is not an option, AFAIK RPi will need at least kernel/u-Boot on SD card. Partitioning would be tricky and would say we need to test few possible setups and compare them :-)

MTRichards, 2016-01-02:
When I installed on the Raspberry Pi 2 with the disk ready to go, the BerryBoot option helped a ton.
http://www.berryterminal.com/doku.php/berryboot

AllI had to do was point and click, and it downloaded the image I wanted to the card. Maybe it is possible to create an ownCloud berry boot image?

enoch85, 2016-01-06:
Ubuntu snappy makes 4 partitions of the SD card, I don't get it yet. But I will keep investigating.

miska, 2016-01-10:
So here is my setup:
http://michal.hrusecky.net/2016/01/my-way-of-booting-pidrive-raspberry-pi2/

enoch85, 2016-01-14:
Thanks @miska ! Great blog btw.

oparoz, 2016-01-20:
I vote for keeping the OS on the SD card and the data on the HDD.

Each OS should make a backup of its config files to a reserved partition on the HDD, so that users can quickly switch OS and restore settings.

jospoortvliet, 2016-01-21:
@oparoz I think that is great, but be sure to put frequently changing files on the hard drive: that is, especially, the database. The normal OS files are probably fine, they're not written to that often if you do the proper tmpfs stuff for /tmp and such.

oparoz, 2016-05-05:
* OS is on SD card
* /tmp is in RAM
* ownCloud files and DB are on the HD

closed 2016-05-05 by oparoz