owncloud-archive/pi-image #7: Getting Groupware to Work

Offerel, 2016-01-10:
Hi together,

over the last days i do not have so much time, but i will share you way to bring the PiDrive to life. My primary goal ist it, to have this thing as a "Homeserver" for the whole family. My most important goal was it, to have all the important data on it. Since i have used a Groupware called "Zarafa" long time ago, i had the idea, to bring this to live on the little Pi. For now, it works with a really good performance for 5 Users on my Network. I didn't used the image which was in the Box for some reasons. First was it, that i have tried to compile Zarafa from Source under Raspbian but this was a horrible Experience. After googling a little bit, i have found ready made Pakets for a Linux Distribution called Arch Linux. So i installed Arch Linux and the Zarafa Pakets. With the first setup, i couldn't successfully setup the Zarafa Server. Reason was the SD Card. The Card is to slow to initially setup the needed MySQL DB, so i ended up with a corrupted installation.

With my next try, i added the HDD to the setup, formatted it with ext4 and now the folders where the System need to write and read a lot, i have created on the HDD. I set than some symlinks to the old paths to get not in trouble and this time the setup was successfully. Zarafa was running, i could create some user-stores and migrated a bunch of my old data to the Server (round about 1GB E-Mail Data, 10.000 Contacts and some Calendar Entrys). This took some time, since i couldn't use the imapsync Tool which seems to no be a free tool anymore.

The next thing was the Mailserver. I setup it up only for local delivery and all external Mails are send via my old Provider. So no one can use it as a Spam Relay. Then i secured the whole Server with SSL Certificates for my own Domain from StartSSL.com.

Since now 1 Week the Server runs very smoothly for this users. All Smartphones could use the Server via Exchange ActiveSync out of the Box to Sync Mails, Contacts, Calendar, Taks and Notes. The same is for the Maiclients from Windows 8.1, Windows 10, eM Client and Outlook starting from Outlook 2013. Outlook 2010 and earlier versions could connect to the Server via IMAP for Mails and CalDAV and CardDAV for Calendar and Contacts. A small WebDAV implementation does work also (which is only a little hack by me), but i could use it for file Backups and Syncing some FIles to Smartphones and more. Its based on SabreDAV like ownCloud to.

A little Samba Share is also running, to use the PiDrive as a Datagrave for my Music, E-Books, Picture, Videos and more. a minidlna Server uses this as a Library, so that i could stream this Content to all DLNA Clients in the House. 

Performance of the SMB Share could be better. Its a little bit faster than on the the NAS on the Router. I come up with Data between 13 - 25MB/sec via LAN and 9MB via WLAN. A HDD on a FritzBox 7490 on the same Network comes to 10 -12 MB/sec. I think the USB /Network Chip on the PI is limiting here.

I didn't have installed OwnCloud now, but it seems that is a simple Task. Next step would be, to bring this setup to easy to install image, which should make the setup easy for a Enduser. I don't know if the Details are interesting enough for the Rest of you. If so i could make HowTo or Tutorial.

BTW: All the Installation was at this on a headless install, meaning, i haven't attached a Monitor on the PI. All was running via SSH or Browser (After installing Webmin).

miska, 2016-01-10:
Just a suggestion, ownCloud has Mail app, Contacts app, Calendar app and of course files app, so you can probably replace you Groupware with ownCloud, although interface will be different, it might be better integrated. I agree that we should have functions like these in the final product, but I would go for ownCloud as it has the functionality and this is supposed to be ownCloud device...

Offerel, 2016-01-10:
If I'm right, OwnCloud is very limited in this point. I didn't see any way to manage Meeting request as some of the simplest form of Groupware Function. No ActiveSync Implementation ans some other things I miss really which are important for me. Because of this important need a more complex Groupware Functionality. That's why that's why I choose Zarafa.

oparoz, 2016-01-10:
@Offerel is correct, the Mail app is purely a webmail client. The problem with shipping a Groupware is that you need to include the full email stack...and manage it. Who will support it when things break? The goal of this device is to make it easy for newbies to start using ownCloud without having to worry about things like maintenance of the OS or other apps.
I still think there is room for add-ons such as Zarafa, Piratebox, any-torrent-app, etc. but they can come later, as separate projects, with their own support teams, etc.

Offerel, 2016-01-11:
Maybe we can add such things via a "AddOns" - Function?

oparoz, 2016-05-05:
Let's close this since that can be implemented via individual snaps if a user wants Zarafa alongside ownCloud.

closed 2016-05-05 by oparoz