owncloud-archive/pi-image #12: Large File Sizes over 2GB on a 32bit System.

Kawohl, 2016-01-13:
When i ran owncloud on my orangepi,  i couldn't upload files larger than 2GB because php on 32bit systems can't handle files larger than that by default.... . 

There are workarounds somewhere....

I think we should make shure the owncloud-pi can handle larger files.

oparoz, 2016-01-13:
According to the documentation, 2GB is going to be a hard limit: https://doc.owncloud.org/server/9.0/admin_manual/configuration_files/big_file_upload_configuration.html

Kawohl, 2016-01-19:
I think this is a major issue...... . 
Using owncloud as a base fileserver for family enviroments( or similar ) the 32bit and the ceiling of 2GB  will cause a lot of frustration. Any folder or file larger than 2GB will fail.......

MTRichards, 2016-01-19:
Doesn't the sync client get around this with chunking, or does the assembly also hit the 2GB limit?

oparoz, 2016-01-20:
> Doesn't the sync client get around this with chunking, or does the assembly also hit the 2GB limit?

Correct

jospoortvliet, 2016-03-14:
Good news is that the Raspberry Pi 3 is 64 bit, so I think we should not put too much effort in solving this - by the time we ship a real big batch, the problem isn't half as relevant as it is today. People will have the option of getting the 64 bit version by then...

halfsight420, 2016-04-03:
Adding a side note to this. I got my ownCloud Pioneering drive. It looks like the php.ini file is inaccessible. It shows no permissions on the admin panel. This has left me with a 2 mb upload limit, nonadjustable. I could not get the partition to remount with rw either.

This is due to the file being on the ro mount. I think for easier use this file should be accessible and editable on a future release.

oparoz, 2016-04-12:
@halfsight420 - This will be taken care of, you will have full control over those settings.