owncloud-archive/pi-image #8: Find and setup the device on the local network

jospoortvliet, 2016-01-13:
Let's use this issue to track progress here. Feel free to edit it!

# End goal
you plug the device in the router and you can visit it on a predefined, human-readable address like myname.own.cloud or so.
Getting there should be as easy as possible. Ideally, plug the device in and visit a defined address to finish the configuration. If that is not reliably possible, perhaps connect the device over network (or usb???) with your laptop, visit a predefined address and configure it from there.

# todo
* [ ] find device over the network
* [ ] what else?

Offerel, 2016-01-13:
If im right, there is no way to find the Box outside the local net, if it is plugged in on the Router which is from my site a very good thing.

If the device connects to a Router, it gets a IP Address via DHCP from the Routers DHCP Range, it depends on the Router which Class C Address this is. You can find this Address on the Router Website mostly on the Startpage.

Alternatively you can plug the Device via LAN directly on your Laptop. In this case there it can get

a) also a IP from the DHCP on the Router if the Laptop is also connected additionally to the Router (maybe via WLAN)
b) if it cant get a DHCP IP, than it gets a predefined IP Address from the setup on the Image. In this case, you must define a static IP Address in the same subnet for the Laptop to access the Device.

c) You can have a pre-configured VPN connection to the Device. In this case, there can be a Companion App for Smartphones like Android or the PC to reach and configure the Device.

So in my opinion best solution is case a), you should connect it to the Router and call the DHCP IP in the Browser. Later you can do Portforwarding Pakets via Router to the Device to find it other the Internet directly. Maybe a dedicated App can also help to get it to work, make some changes and manage the Device. There are several Apps for the Pi in the Play Store available. Most of them work via SSH on the local Network or via VPN from external.

miska, 2016-01-13:
Well, it should be in theory possible to make the device locatable via avahi. Wouldn't work for everybody, but would work for some people. Maybe samba has something similar.

We can do both - DHCP and static, but then it might collide with something already on the network.

Apart from that, I think we can display IP address on LCD if connected, that's what I would like to do at first (just because it is easy to do and I almost have it done, not because I think it is that useful).

Offerel, 2016-01-13:
Will there be a LCD Panel included in the Final Box?

miska, 2016-01-13:
There is HDMI which is pointing out of the box. I meant that LCD.

Offerel, 2016-01-13:
Ahh ok. My didnt have seen any other cable as the LAN and USB for Power for now :-) All setup was done via SSH. But you are right, HDMI is a way. But maybe a small LCD Panel can do the trick like on this page: http://pmb.neongrit.net/blog/?p=64

miska, 2016-01-13:
I had in my box HDMI L connector to repoint the HDMI in right direction.

oparoz, 2016-01-13:
* Pick an IP from the network. Like 222. If the network is 10.0.0.1, then the IP would be 10.0.0.222
* Give instructions to users on how to get the IP from the mac address, something like connect the device and do a  `arp -a` on Windows

cmacq2, 2016-01-15:
In the context of Debian something like this in `/etc/network/interfaces` would suffice:

```
# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
allow-hotplug eth0
iface eth0 inet dhcp
iface eth0 inet6 dhcp
```

Combined with an Avahi config for publishing the R.Pi2 via m-DNS/DNS-SD...

Note that with IPv6 you have link local addresses which you can generate from the eth0 MAC. Now it may be the case the MAC isn't set by default on the Pi, in which case a script on boot using `ethtool` or something can fix that.

So then all you have to give to the end-user is a simple HTML page with a bit of embedded JavaScript that contains a form for inputting the MAC, and redirects to the R.Pi based on that. (You could go beyond that even, like testing 'the device is available' and so on...)

miska, 2016-01-17:
Got my image to display IP on attached monitor:
http://michal.hrusecky.net/2016/01/getting-to-your-pidrive/

LEDfan, 2016-01-18:
I was just thinking this could also be a way to automatically set up the network:
 
 1. connect the PI to the LAN via UTP
 2. set DHCP config (i.e. https://github.com/owncloud/pi-image/issues/8#issuecomment-171934122)
 3. start a Qt/C++ program on your computer (very simple, cross platform). Which is just an executable/portable exe on Windows.
 4. This program does a network scan via [nmap](https://nmap.org/). And filters on a RPI hostname. This requires first a "ipconfig" or "ip addr" command to find out the IP subnet and netmask. If we use Qt this can easily be done with:
````
#include <QtNetwork>
#include <iostream>
int main(int argc,char** argv) {
      QNetworkInterface* network = new QNetworkInterface();            
      foreach(QHostAddress addr,network->allAddresses()) {
	std::cout<<addr.toString().toStdString()<<std::endl;
      }
}
````
The only disadvantage of Qt is that it's a large exe on Windows.
Note: I first wanted to search on MAC but then the RPI needs a vendor specific MAC address + this requires root privileges on Linux.
4. Or you have to type the MAC address printed on the box of the RPI.
5. you now know the IP of the RPI

Note that for port forwarding etc a static ip address is needed. Some providers (e.g. Telenet in Belgium) uses random 192.168.0.x addresses as DHCP lease. There isn't a pool for static addresses, so it could be possible that you set up an static address and there is an IP address conflict. 

This could be interesting too: https://github.com/j-keck/lsleases

LEDfan, 2016-01-18:
Nmap can output directly to xml: `nmap -sn -oX test.xml 192.168.0.0/24`. `Test.xml` :
````
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE nmaprun>
<?xml-stylesheet href="file:///usr/bin/../share/nmap/nmap.xsl" type="text/xsl"?>
<!-- Nmap 7.01 scan initiated Mon Jan 18 09:22:33 2016 as: nmap -sn -oX test.xml 192.168.0.0/24 -->
<nmaprun scanner="nmap" args="nmap -sn -oX test.xml 192.168.0.0/24" start="1453105353" startstr="Mon Jan 18 09:22:33 2016" version="7.01" xmloutputversion="1.04">
<verbose level="0"/>
<debugging level="0"/>
<host><status state="up" reason="conn-refused" reason_ttl="0"/>
<address addr="192.168.0.1" addrtype="ipv4"/>
<hostnames>
</hostnames>
<times srtt="663" rttvar="5000" to="100000"/>
</host>
<host><status state="up" reason="conn-refused" reason_ttl="0"/>
<address addr="192.168.0.60" addrtype="ipv4"/>
<hostnames>
</hostnames>
<times srtt="322" rttvar="3764" to="100000"/>
</host>
<host><status state="up" reason="syn-ack" reason_ttl="0"/>
<address addr="192.168.0.205" addrtype="ipv4"/>
<hostnames>
</hostnames>
<times srtt="529" rttvar="5000" to="100000"/>
</host>
<host><status state="up" reason="conn-refused" reason_ttl="0"/>
<address addr="192.168.0.206" addrtype="ipv4"/>
<hostnames>
<hostname name="hostname" type="PTR"/>
</hostnames>
<times srtt="38" rttvar="5000" to="100000"/>
</host>
<runstats><finished time="1453105368" timestr="Mon Jan 18 09:22:48 2016" elapsed="15.41" summary="Nmap done at Mon Jan 18 09:22:48 2016; 256 IP addresses (4 hosts up) scanned in 15.41 seconds" exit="success"/><hosts up="4" down="252" total="256"/>
</runstats>
</nmaprun>

````

oparoz, 2016-03-21:
With the current image, `owncloud.local` doesn't work, but after matching the mac to an IP, it's possible to log in using the IP.
I'm not surprised as `. local` isn't used on my network, but we need a better solution :)