owncloud-archive/pi-image #13: Ubuntu Snappy

oparoz, 2016-01-15:
Pro: 
* The web server, PHP and ownCloud are built like an app, and the maintainer (@enoch85) handles the build to keep it updated with the latest versions. The update command is very easy, just `snap update`. The update process could be automated as well.
* Built on Ubuntu which is one of the largest communities around.
* Good performance (source?)
* Snappy apps and Snappy Ubuntu Core itself can be upgraded atomically and rolled back if needed. There are issues, see cons
* [Ease of core image installation](https://developer.ubuntu.com/en/snappy/start/raspberry-pi-2/) (ubuntu/win/mac) 
* Built on Snappy Ubuntu Core, ownCloud would run completely confined, limiting the damage if it's compromised. That's from a security perspective
* From a packaging perspective, Snappy Ubuntu Core uses .snaps, which include all their dependencies for rock solid stability.
* Updates are transactional

Con:
* Doesn't have apt-get so the end-user have to rely on that the maintainer does his job to keep updated with the latest ownCloud version
* It requires learning and using a new packaging type. Not for the end-user though
* Very new (~1 year). Has not been extensively tested
* Unstable, especially on Pi2
* Almost nobody in the team has used it

jospoortvliet, 2016-01-21:
Here's some interesting stuff going on: https://github.com/kyrofa/owncloud-snap ;-)

oparoz, 2016-01-21:
Yep, I think it will be a great image once Snappy Core 16.04 is released

enoch85, 2016-01-21:
Build is about to be uploaded to the store, if you guys want to try it out. Needs some more polishing though.

kyrofa, 2016-01-27:
Alright, this has been uploaded. Indeed, needs some more work, but it'll give you the right idea anyway.

oparoz, 2016-01-27:
Since 16.04 images [are already available](https://people.canonical.com/~mvo/all-snaps/). Why not build on top of them? That would mean people choosing that image in March-April would be able to update to something pretty stable right after the first installation.

kyrofa, 2016-01-27:
@oparoz as soon as all-snaps is merged into rolling (16.04) I'll release the .snap for that as well.

jospoortvliet, 2016-03-14:
So agreed with the building on 16.04 images for the first batch!

We can hopefully ship the final for the Raspberry Pi 3, that even takes away the problem with the file size limitation on 32bit systems - whoehoe!

kyrofa, 2016-03-14:
@jospoortvliet note that the rpi3, while having 64-bit hardware, does not run 64-bit right now.

oparoz, 2016-03-14:
Yeah, only a handful of Pi distributions run in 64-bit.

So, ideally, the build tools allowing us to generate those 2 images should be uploaded on here so that people can start adding scripts.

kyrofa, 2016-03-14:
@oparoz what scripts are we talking about, exactly? Perhaps they should be snaps as well.

oparoz, 2016-03-14:
@kyrofa: Scripts which achieve this goal per example: https://github.com/owncloud/pi-image/issues/8

oparoz, 2016-03-14:
It's fine if they are snaps, but we need to be able to generate images which include everything in order to tests the final result.