owncloud-archive/pi-image #15: PHP stack

oparoz, 2016-01-15:
### Required configuration
* pnctl
* intl

### List of required modules
* redis
* apcu?

### PHP version

5.6 or 7? It will depend on the availability of modules

LEDfan, 2016-01-17:
PHP 7 would be awesome for the performance, but there are still some issues with it: 
 - imagemick must be compiled by yourself (not difficult), IIRC it's needed for all supported MIME types of gallery? (I guess you know it :P)
 - sabre dav isn't 100% working on PHP 7, so calendar isn't working. Not sure if this is already fixed
 - not officially supported by the distro's
 - maybe it contains some bugs, which will need a regular upgrade to a new minor version?

I'm not sure if running redis on a RPI will improve the performance of ownCloud since the low amount of RAM, which means there would be less RAM for MySQL AFAIK. 

Another thing that came in my mind is the InooDB buffer pool size which should be correctly set, because otherwise MySQL/MariaDB can crash if there are a lot of requests. (E.g. with the gallery app and a bunch of images) Ref: http://dba.stackexchange.com/questions/27328/how-large-should-be-mysql-innodb-buffer-pool-size

LEDfan, 2016-01-17:
For choosing a PHP version it's important that it's long supported by ownCloud. I.e. choosing for PHP 5.4 now isn't a good idea since the latest version of Sabre-dav needs at least 5.5

tflidd, 2016-01-17:
> I'm not sure if running redis on a RPI will improve the performance of ownCloud since the low amount of RAM, which means there would be less RAM for MySQL AFAIK.

I did some upload performance tests in December and mysql is a bottle neck (i/o-wait). redis should reduce the database load but it didn't really help: https://github.com/owncloud/core/issues/20967 (perhaps you need to tune more parameters).

PHP7:
> not officially supported by the distro's

Even if it's not a big deal to compile modules, how should this be maintained? dotdeb.org provides some packages for PHP 7 but redis support is still missing (I'm not sure about imagemick).

LEDfan, 2016-01-17:
> Even if it's not a big deal to compile modules, how should this be maintained? dotdeb.org provides some packages for PHP 7 but redis support is still missing (I'm not sure about imagemick).

That  was indeed what I wanted to tell :). I think redis-php isn't even available as precompiled module and thus must be compiled always by the user. I think compiling modules on the RPI without the user knowing what's happening is impossible. A simple `./configure` will stop the compilation when a dependency isn't found.

LEDfan, 2016-01-17:
Plus, compiling on RPI will also be not very fast...

oparoz, 2016-01-17:
All good points. I think PHP 7 should then be considered too young for an arm based project to be released in Q1. Compiling modules ourselves is not a problem, but it would increase the maintenance work load and imho we should leave that to the distros and focus on configuring the OS+stack.

Regarding Redis, it really helps when simply browsing files, because mySQL based file locking offers really bad performance. We'll just have to find the sweet spot in terms of allocated memory.

jospoortvliet, 2016-03-14:
With ownCloud 9.0 out and PHP 7 maturing quickly, I think we should go for it. It provides a performance boost we can use and if there are problems with it in the first batch - the people who get one can help shake those out ;-)

oparoz, 2016-03-14:
Redis and imagemagick are available, so that should be OK. I'm sure there are still lots of segfaults issues, but processes are not long running, so not much of a problem.