owncloud-archive/pi-image #17: Requirements for a base OS

oparoz, 2016-01-22:
## Must have 

### First launch
* Creates required partitions if they don't exist
* Registers on the network
* Installs ownCloud (if it doesn't exist)
* Configures ownCloud
* Sets up a welcome page (www)
* Configures SSHd
* Configures cronjobs

### Features
* Boots from SD card
* Can be updated without re-flashing
* Backed by large community. That makes it easier to find answers if users wants to customise their installation
* Doesn't **require** custom package creation (only configuration). This is to void long term maintenance issues
* Stable, proven branch of the OS (as opposed to experimental, dev previews, etc.)

### Partitions
The data has to be kept separate from the OS in order to avoid lock-ins and data loss, but also so that users can try various images before settling in or leave one when it becomes unmaintained/unstable.
Write heavy partitions have to be stored on the hard drive.

#### SD card
* /
* /etc
* /var

#### HDD
* /home
* /sql
* /www
* /var/log
* /backups

## Nice to have

### First launch
* Shows IP address over HDMI
* Shows initial account info over HDMI

enoch85, 2016-01-23:
Nice summary @oparoz :+1:

oparoz, 2016-01-23:
Thanks, it's a start :D.
I'm hoping for participants to complete it and have discussions on points which they don't agree with so that we can define the global architecture to make images sort of compatible. That way, end users will be free to move around and experiment things without having to start from scratch every time.

ezraholm50, 2016-02-06:
https://github.com/ezraholm50/BerryCloud

oparoz, 2016-02-07:
I couldn't find the part where the hard drive is used for data, backups and swap while leaving the rest on the SD card.

ezraholm50, 2016-02-07:
Usbhd.sh contains the code for that, the script needs some work give me 2 days. Its quite easy just dd mmcblk0p2 to sda1 and set it to boot in fstab and reboot. Also did some cleaning up in the setup.sh...

ezraholm50, 2016-02-07:
Damn, no edit button on android.  In that setup we wont use the root sd card partition anymore, if we want to use your version, data, swap and backup that will be easier.

How about a swap partition instead of a file?

oparoz, 2016-02-07:
Swap partition is best. I've updated the partition requirements as I think we can use folders in `/home` instead of individual partitions, but I admit I'm more used to ZFS where folders can be treated as partitions and tuned individually, so anybody with Linux tuning experience can suggest improvements.

ezraholm50, 2016-02-10:
@oparoz 
Have a look at my usbhd.sh script, just ran the script and it works :dancer: 
You can choose if you want to use the HD as root partition, it also sets up a 2 gb swap partition.
Else, if answered no, a swap file is created on the SD card.

oparoz, 2016-02-10:
>The whole ROOT partition is on the HDD (SSD in my case), would that be a problem?

Yes, I think that's the biggest hurdle, because if every image does the same, then users lose all their data each time they install a new OS, because the partitioning scheme can be very different from image to image.
The HD should only be used for permanent (/home) or write-heavy (/tmp) partitions. Swap is a special case. I think it's best  to have it on the HDD.
In my mind, the OS would already be on the SD card, it creates the /home partition if it doesn't exist (the other partitons as well), then boots, self-updates and launches the server.
If home already exists, that means it meets the specification and the ownCloud config file can be trusted to point to valid directories and a working database.
Does that make sense?

cryptworks, 2016-02-25:
I would second oparoz on the suggestion of moving the write intensive processes over to the HDD but leaving the root filesystem on the SD card.