owncloud-archive/pi-image #18: Overclocking

ezraholm50, 2016-01-25:
Hey guys, I would like to know your thoughts about overclocking the RPI.
Next I will share my findings.

OS: Ubuntu Core 15.04
Hardware: RPI2B

The following settings void warranty. I have been using this setup for over 2 months, its runs smooth without any errors. It does seriously make a difference running for example apt-get update & apt-get upgrade and other general load times of the front end. I will try to post the benchmarks with and without later on. I do have 3 heat sinks installed on the RPI2B and a CPU Fan.

These are the most extreme settings I could get out of the RPI2, with higher values it breaks the OS and I had to do a reinstall of the img on the sdcard.

Settings:
gpu_mem=16
cma_lwm=16
cma_hwm=32
boot_delay=1
force_turbo=1
initial_turbo=30
temp_limit=80
arm_freq=1100
core_freq=600
sdram_freq=500
over_voltage=8
over_voltage_sdram=4
over_voltage_sdram_c=4
over_voltage_sdram_i=4
over_voltage_sdram_p=4

A safer and better alternative is to use 'raspi-config' which also allows you to do various other useful things, which of course could also be automated...
https://github.com/asb/raspi-config
https://www.raspberrypi.org/documentation/configuration/raspi-config.md

A small indication:

[image]

Setup Options

    1 Expand Filesystem              Ensures that all of the SD card storage is available to the OS
    2 Change User Password           Change password for the default user (pi)
    3 Enable Boot to Desktop/Scratch Choose whether to boot into a desktop environment, CLI
    4 Internationalization Options   Set up language and regional settings to match your location
    5 Enable Camera                  Enable this Pi to work with the Raspberry Pi Camera
    6 Add to Rastrack                Add this Pi to the online Raspberry Pi Map (Rastrack)
    7 Overclock                      Configure overclocking for your Pi
    8 Advanced Options               Configure advanced settings
    9 About `raspi-config`           Information about this configuration tool

# The overclocking options here do not void warranty and do speed up the RPI2B.
These are the options:

################################################################################
Overclocking Settings

ARM, SDRAM and GPU each have their own PLLs and can have unrelated
frequencies.
The GPU core, h264, v3d and isp share a PLL, so need to have related
frequencies.
pll_freq = floor(2400 / (2 * core_freq)) * (2 * core_freq)
gpu_freq = pll_freq / [even number]

The effective gpu_freq is automatically rounded to nearest even integer, so
asking for core_freq = 500 and gpu_freq = 300 will result in divisor of
2000/300 = 6.666 => 6 and so 333.33MHz.

Standard Profiles:
                 arm_freq    core_freq    sdram_freq    over_voltage
    -------------------------------------------------------------------------
    None            700            250             400              0
    Modest          800            300             400              0
     Medium          900            333             450              2
     High            950            450             450              6
     Turbo           1000           500             500              6

################################################################################

# These options (not default) are from the Ubuntu Mate 15.04 OS and explain a bit more about all the different options

https://github.com/ezraholm50/BerryCloud/blob/master/Ubuntu-Mate-15.04-config.txt

tflidd, 2016-02-12:
The moderate overclocking, which doesn't void the warranty, could be interesting. Do you have any numbers on the performance improvement? Does it help during heavy file operations?

ezraholm50, 2016-03-22:
Sorry for the delayed reply, been busy. 

No hard numbers, only thing I have is when the overclocking is enabled it saves me 20 minutes or more installation time (My BerryCloud repo) of total 60 mins without overclocking.

Not that into benchmarking etc yet so its hard for me to get some solid figures posted.
Could use some help on this one.