owncloud-archive/pi-image #22: Which ownCloud apps to ship, which to enable

oparoz, 2016-03-19:
Which apps should be shipped and which one should be enabled?

### Requirements

* ownCloud 9 compatible
* PHP 5.4 compatible

`core` uses ancient libraries, so modern app may not run properly if they use libraries designed for more recent versions of PHP.

### Apps

| App 	| Shipped 	| Enabled 	|
|:---:	|:-------:	|:-------:	|
|   Move	|    x	|       x	|
|    Sharelinks	|       	|         	|
|   Calendar	|         	|         	|
|   Contacts	|         	|         	|
|  Tasks	|         	|         	|

I'll let @miska add his apps :)

enoch85, 2016-03-19:
The snappy is with all the latest stuff, PHP 7, MySQL 5.7 and so on. You mean this will be an issue @oparoz ?

oparoz, 2016-03-19:
@enoch85 - PHP 7 is only an issue if we start to include cutting-edge apps to do some very specific things for this project. To give you an example, you need a new version of PHPUnit to test on PHP7. In this case it only affects testing and devs can find workarounds, but let's say you need the latest version of GuzzleHttp to make your life easier. That will break ownCloud as an older version is already included in the 3rd party folder.

ezraholm50, 2016-03-21:
My 2 cents, keep it simple and only enable: share links, share files, documents (requires libreoffice), gallery, mail, notifications, pdf viewer, text editor, versions, update notification, OC IPv6.

We should also install libre office whether we ship and/or enable documents.

Then the users can install/enable other apps to their liking. Also, enable or disable experimental appstore?

oparoz, 2016-03-21:
>enable: ...mail,...

I wouldn't enable mail as it only makes sense if you use the other "groupware" apps. It also doesn't store anything locally which can be confusing for newcomers.
What we should do is have a nice page in the manual explaining how users can keep their appointments, contacts, tasks and notes synced with all their devices and how the mail app can use all this information when reading and sending mail. Let's not forget that mail is currently far from realising this vision as it's only at v0.4

>documents (requires libreoffice)

Maybe for phase 2 as this might be too resource hungry, no?

The rest seem reasonable to me.

>Also, enable or disable experimental appstore?

I'd say,no. Too many things can go wrong... Advanced users will know what to do.

>Edit: how about ClamAv with Files antivirus?

That thing kills standard servers, so I don't think it's a good candidate for a Pi2, no?

tflidd, 2016-03-22:
> My 2 cents, keep it simple and only enable: share links, share files, documents (requires libreoffice), gallery, mail, notifications, pdf viewer, text editor, versions, update notification, OC IPv6.

I agree with @oparoz to use a very basic selection of apps and don't ship mail and documents.

What can RPI2 be used for and what can the device handle?
* sharing files
* sharing calendar/contacts
* sharing pictures (gallery app)
* sharing other media (music, video)
* other stuff

I would start at the top of the list, and even don't ship the gallery by default if we can't get it working with reasonable performance (suitable php7-modules). We should keep in mind that users might want to enable an app and owncloud shouldn't already exhaust all resources.

It would be interesting to know what combinations of apps can run (for testing).

oparoz, 2016-03-22:
>I would start at the top of the list, and even don't ship the gallery by default if we can't get it working with reasonable performance (suitable php7-modules). 

We can't really do that as it's part of the base distribution and because it's the app providing the large previews (maybe one day the slideshow will be a separate app...), but there isn't really a performance issue anyway. It's not much worse than using Files. Thumbnails have to be generated by one or the other. 

Based on the target market, I wouldn't even enable calendar and contacts in the first version since the device will probably not be accessible from the outside world, unless these apps can act as aggregators and collect data from external services?

What will people use the device for?
* Share pictures and videos
* Share files

All on their home network, between family members or office workers.
Having office documents previews would probably be useful, but we I'm worried about the memory requirements of LibreOffice and its impact on the system every time a preview has to be generated... Ideally, there should be a snap people can load and they can then enable the previews for those files and see if it works for them.

Once we connect the device to external proxys, we can then enable more apps and publish how-tos.

ezraholm50, 2016-03-22:
> enable: ...mail,...
> I wouldn't enable mail as it only makes sense if you use the other "groupware" apps. It also doesn't store anything locally which can be confusing for newcomers.
> What we should [...]

Your right about Mail indeed... my bad.

As for Clamav, if only used for uploading i barley noticed any difference, with an overclocked RPI.
But if this would indeed be a build just for home users, we should leave it out for a while and review its need later on.

As for Documents, here i also see no lag or performance loss, using libre-office -no recommends. (Only thing is I'm not able to simulate around 5 to 10 users using OC at the same time. )
And its also a welcome feature within ownCloud that users will use, imo.
In combination with redis cache I don't think this is a problem.

As for gallery, there's no way to not have it, imo. Its one of the pillars of OC after the file storage/sharing.

Finally I do think we should overclock to the settings that keep warranty intact, I have made a post in the issues here about it. Also my BerryCloud repo contains some info.

And @oparoz what to do with SWAP, partition/file, size, swappiness?
My build holds a file of 2GB at 60 swappiness, im not that into swap to say whats best.

oparoz, 2016-03-22:
>As for Clamav, if only used for uploading i barley noticed any difference, with an overclocked RPI.

That's good to know :), but let's not forget that most people won't overclock their hardware.

>As for Documents, here i also see no lag or performance loss

OK, something to test then. We need to figure out how to get LibreOffice as a snap and communicate with the binaries
@kyrofa - Any ideas?

>Finally I do think we should overclock to the settings that keep warranty intact, I have made a post in the issues here about it. 

I don't know what to think... The kit runs super hot already, but maybe that's because the case doesn't act as a heattsink. Maybe if the final design has better cooling properties it could be doable.
@ezraholm50 - Did you graph the temperatures you get in overclocked mode?

>And @oparoz what to do with SWAP, partition/file, size, swappiness?

I have absolutely no idea :D 
There is **zero** swap in the pi-image.

ezraholm50, 2016-03-22:
```

As for Clamav, if only used for uploading i barley noticed any difference, with an overclocked RPI.
That's good to know :), but let's not forget that most people won't overclock their hardware.

As for Documents, here i also see no lag or performance loss
OK, something to test then. We need to figure out how to get LibreOffice as a snap and communicate with the binaries
@kyrofa - Any ideas?

Finally I do think we should overclock to the settings that keep warranty intact, I have made a post in the issues here about it.
I don't know what to think... The kit runs super hot already, but maybe that's because the case doesn't act as a heattsink. Maybe if the final design has better cooling properties it could be doable.
@ezraholm50 - Did you graph the temperatures you get in overclocked mode?

And @oparoz what to do with SWAP, partition/file, size, swappiness?
I have absolutely no idea :D 
There is zero swap in the pi-image.
```

Indeed well just investigate some more and then vote about the overclocking, in the mean time I will try to get some hard stats on the overclocking. Might need some help...
With overclocking there is an option to set the degree, if that is passed it reverts the overclock settings to default, as a build in safety option.

My kit has 3 heatsinks on the rpi and a little fan, that could be an option for the final product.

tflidd, 2016-03-22:
Are there standard-procedures that can be used for a benchmark?

oparoz, 2016-03-22:
`ab` is one

https://httpd.apache.org/docs/2.4/programs/ab.html

ezraholm50, 2016-03-22:
Also we might want webmin?

jospoortvliet, 2016-03-23:
Calendar and Contacts are core reasons for people to get ownCloud. I think we should ship AND enable them, just to give a complete user experience.

IPv6 and ownCloud Proxy should be there, too - at some point we need an app which guides people through the setup of their network but for now, these apps proxy for that.

oparoz, 2016-03-24:
I think Proxy can be shipped, but shouldn't be enabled yet. The image has to get stable first.

oparoz, 2016-03-24:
To Gallery+ or not to Gallery+? Zhat is ze questionn...

V15, to be released this weekend at the latest, features performance and usability enhancements which will improve the experience of users of low-powered hardware.
Since most beginners stick with the default, they may not even realise they can get a better version from the app store and reject the whole experience, thinking the whole thing just doesn't work.
If we stick with Gallery, they won't get the new features until July at the earliest.

ezraholm50, 2016-03-24:
We can leave the experimental store disabled and enable Gal+ right? don't see the problem after testing?

jospoortvliet, 2016-03-30:
Oparoz, about the Gallery - would generating thumbs in a cron job not help A LOT? Perhaps that should be a prio for this then... ?

enoch85, 2016-03-30:
@jospoortvliet 
> Calendar and Contacts are core reasons for people to get ownCloud. I think we should ship AND enable them, just to give a complete user experience.

Agreed. +1

@oparoz 
> To Gallery+ or not to Gallery+? Zhat is ze questionn...

Gallery+

oparoz, 2016-03-30:
>Oparoz, about the Gallery - would generating thumbs in a cron job not help A LOT? Perhaps that should be a prio for this then... ?

https://github.com/owncloud/pi-image/issues/28

oparoz, 2016-04-13:
There are currently 2 issues

1. We need to modify the way ownCloud is installed by the snap in order to be able to enable apps before the user first logs in
2. ownCloud marks any app which isn't downloaded from the app store as experimental, which means we either don't ship any extra app or mark them as "official" ...