owncloud-archive/pi-image #23: We need instructions on how to flash the images from Windows

oparoz, 2016-03-21:
Some sort of image writing how-to should be added to the wiki.

### How to connect the hardware to a PC

I think we just need to let people know they can connect the drive to a USB3 port to write a new image to disk.

### How to flash the images

We have instructions for Linux, but should have them Windows and OSX as well

1. Recommend tools to uncompress the downloaded images
1. Recommend USB flashing tools which work with these images
1. Document how to both write and expand

----
@kyrofa @enoch85 @jospoortvliet

oparoz, 2016-03-21:
The image resizing part is problematic... It would be best if performed during the first boot.

ezraholm50, 2016-03-21:
Have a look at my berrycloud readme for the windows image stuff. Linked to
the raspberry site. Also the resize is easy to fix. Example add to rc.local
remove if successful

oparoz, 2016-03-21:
## Hard drive

### Preparation

1. Connect your HD to both a USB3 port for data and USB2 port for power (via a micro-USB adapter/cable)
1. Extract the writable.img.xz file using ??? (Winrar works)

### Writing the image

1. Download Rufus from https://rufus.akeo.ie/
1. Launch Rufus
1. Click on "Advanced options"
1. Select your HD under "Device". Mine was labelled "NO_LABEL (Disk 1) [1TB]
1. Under "Create a bootable disk using", click on the CD-ROM icon next to it
1. Switch the file type filter from ISO to "* all files"
1. Select the extracted writable.img which was prepared earlier
1. Click on `Start` and wait a few seconds for the image to be written to the HD
1. Click on `Close`

### Expanding the image

### As a home user
1.  Download "Paragon Partition Manager Free"
1. Open the app, don't try to resize from the launcher
1. Select your partition (3.5GB) on the external disk
1. Modify the size to the maximum allowed size
1. Apply your changes

oparoz, 2016-03-21:
@ezraholm50 - Thanks for the info. We can add steps for more than one writer, in case people already have one handy.

>Example add to rc.local remove if successful

Yes, or do a simple test, as people may change their drive sometimes.

kyrofa, 2016-03-21:
> The image resizing part is problematic... It would be best if performed during the first boot.

It actually should be (if it's not, it's an Ubuntu Core bug). I like to do it myself, but Ubuntu Core should resize the writable partition on boot. Feel free to give that a shot.

oparoz, 2016-03-21:
Ah, thanks @kyrofa , so maybe the readme should be updated as I was under the impression that without using `write-and-expand` I would be limited to the 3.5GB partition.

kyrofa, 2016-03-21:
@oparoz oh you're right I did say that didn't I? Okay, let me actually test this and verify that it's working and I'll update it.

oparoz, 2016-05-05:
There is now a wiki entry which can be completed as needed: https://github.com/owncloud/pi-image/wiki/Flashing-the-image

closed 2016-05-05 by oparoz