owncloud-archive/pi-image #24: Caching

oparoz, 2016-03-21:
After having witnessed the load placed on MySQL, I'm hoping it can be reduced via caching.

I'm voting to use Redis as it can be used for file locking as well.

Here are some numbers, courtesy of @Niluge-KiWi

* filelocking.enabled + `memcache.local` APCu + `memcache.locking` Redis: 
 * avatar/name/id: 70ms
 * config: 110ms
 * list: 80ms
 * thumbnails: 115-165ms (5 requests)
 * page ready: 0,95s
 * total: 1,95s
* filelocking.enabled + `memcache.local` Redis + `memcache.locking` Redis: 
 * avatar/name/id: 90ms
 * config: 200ms
 * list: 90ms
 * thumbnails: 130-180ms (5 requests)
 * page ready: 1s
 * total: 2,45s
* filelocking.enabled + `memcache.local` none + `memcache.locking` Redis: 
 * avatar/name/id: 75ms
 * config: 130ms
 * list: 85ms
 * thumbnails: 130-180ms (5 requests)
 * page ready: 1,05s
 * total: 2,45s
* filelocking.enabled + `memcache.local` none + `memcache.locking` none: 
 * avatar/name/id: 75ms
 * config: 460ms
 * list: 1,95s
 * thumbnails: 5-14s (5 requests)
 * page ready: 1,05s
 * total: 35s
* filelocking.disabled + `memcache.local` none + `memcache.locking` none: 
 * avatar/name/id: 80ms
 * config: 130ms
 * list: 75ms
 * thumbnails: 115-150ms (5 requests)
 * page ready: 1,05s
 * total: 1,95s

enoch85, 2016-03-21:
Here is a script that you can use togehter with PHP 7: https://github.com/enoch85/ownCloud-VM/blob/master/production/install-redis-php-7.sh

oparoz, 2016-03-21:
Thanks @enoch85

ezraholm50, 2016-03-21:
Redis +1
Have used it on my BerryCloud project and it works flawless imo (with scripts from enoch85)

tflidd, 2016-03-22:
I did some tests a while ago on raspbian jessie on a RPi 2 (nginx, php5-fpm, maria-db, redis, system+db on sd card, storage on external usb stick):
I found the i/o-performance to be the bottle neck especially caused by the database. File-locking should be done by redis (or disabled).

I also disabled journaling for ext4 which increased the performance. I'm not sure if this is a good idea in general, perhaps we could put the database on a separate partition with journaling turned off (regular backups should be run).

I even tried to put the database on tmpfs but there was no major improvement on the upload of a large number of files (I didn't focus on the response of the web-interface):
https://github.com/owncloud/core/issues/20967
That is the reason, I didn't pursue this any further, suitable backup & restore mechanism would be necessary.

To increase the available RAM, I reduced the shared GPU memory to 16M by adding this to boot.txt:
````
gpu_mem=16
````
Perhaps something similar is possible in ubuntu snappy.

oparoz, 2016-03-22:
>To increase the available RAM, I reduced the shared GPU memory to 16M by adding this to boot.txt:

I was just reading about that... 1GB is really low for ownCloud, so the more we can free the better.

Does HDMI output still work with 16MB?

tflidd, 2016-03-22:
> Does HDMI output still work with 16MB?

Yes. But I don't use a desktop environment.

oparoz, 2016-03-22:
>> Does HDMI output still work with 16MB?

>Yes. But I don't use a desktop environment.

Yeah, it's just so that people can still access the device for emergency work or to get their IP address if they don't want to use the command line.

jospoortvliet, 2016-03-24:
I also created an issue for the 16MB video memory thing: https://github.com/kyrofa/owncloud-snap/issues/18

oparoz, 2016-05-05:
Redis will be used for both file-locking and caching.

closed 2016-05-05 by oparoz