owncloud-archive/pi-image #25: Findings snappy core img

ezraholm50, 2016-03-22:
Don't really know where to put all the seperate findings so ill state them here.
Don't shoot me, I just installed a bunch of apps from the app store
Have not had a great deal of time to spend testing.

@kyrofa @oparoz As for the resize, it checked it at boot and said there was more then 10% free space so it did the resize, properly!

# Overclocking, the overclock settings I had from my project don't really work, unable to boot. So I'm just figuring out what settings do work.

# Booting/shutting down seems a bit slow in comparison to the image I used, probably is because of the overclocking that isn't setup on the snappy kit here just yet. I'll do some test runs with boot time to login. @kyrofa I still don't really know what image I got, snappy core with apt without snappy I guess.

# Overall performance is just fine and amazed me, and redis (or something similar) still needs to be installed. 

# Default enable cron, most unknown users would not know what the heck thats about.

# Maybe install smbclient snap? To mount external shares.

oparoz, 2016-03-22:
@ezraholm50 - You can add your requests here: https://github.com/owncloud/ubuntu-snap/issues. That's the repo which should be used in the future to build our custom snap, once ownCloud members know how to.

>Overall performance is just fine and amazed me, and redis (or something similar) still needs to be installed.

Please try to sync a large folder with the device. Mine completely gives up after a few minutes, especially if I'm logged in via the web GUI at the same time. MySQL shuts down and then the network goes down.

oparoz, 2016-03-22:
>Maybe install smbclient snap? To mount external shares.

Yes, that would be nice to have

ezraholm50, 2016-03-22:
My bad will use that one instead. I will try to upload 1.9 GB in a sec

Im running it on a 32 gb SSD i had lying around covered in dust.

closed 2016-03-22 by ezraholm50