owncloud-archive/pi-image #26: Support for RPI 3

BrianSaia, 2016-03-23:
What is the time-line for Raspberry Pi 3 support if any?

kyrofa, 2016-03-23:
No ETA yet; progress is currently being made on u-boot. I can assure you people are working on it, though :smiley: .

oparoz, 2016-03-24:
@BrianSaia - Don't expect too much improvement though as the limiting factor is currently RAM.

miska, 2016-03-31:
I would say that for PiDrive Pine64 or Odroid C2 would make much more sense as they have a decent network and can have even more RAM if needed for some reason...

jospoortvliet, 2016-04-04:
@miska yes, I've asked WD to look into Odroid C2. It is probably going to be compatible with our case, if not let's see if we can make a change to accommodate it.

The Odroid is our best bet in terms of formfactor, community support and price/performance, I think. The ownCloud image would probably be ready for it the moment a Ubuntu Core image would be released for the Odroid, it just needs to be recompiled to 64 bit ;-)

closed 2018-12-28 by SaltyCybernaut