owncloud-archive/pi-image #28: Gallery strategy for devices with low RAM

oparoz, 2016-03-30:
## 1st use

@jospoortvliet said
>Oparoz, about the Gallery - would generating thumbs in a cron job not help A LOT? Perhaps that should be a prio for this then... ?

I think it would achieve the opposite on this device... Each time the cron job would kick in, the device would die or be brought to its knees and a UI which becomes randomly unresponsive is particularly annoying.

### Possible solutions

* Educating users, but that rarely works
* Showing a message on first use. Will users remember? @jancborchardt 
* Adding a button in settings to generate thumbnails by calling an API. but the process would die when the user leaves the page...
* Adding an occ command, but only advanced users will dare to use that

Anything else?

## After that

### Thumbnails at upload time

We could create thumbnails as they're uploaded, but again, on this category of device with low RAM any PHP request is dangerous for the stability of the whole thing.
I've monitored memory use today and I've noticed that we have roughly 150MB to work with. Using Gallery, memory drops quickly as rows are shown. When it's gone, Apache crashes, Gallery gets a 500 and can't do anything else in the current album. I might be able to do something about that, but if a background process makes Apache crash, then any app will suffer the same fate.
We'll have to move to PHP-FPM before revisiting this strategy for the Pi I think, but that's the path Gallery is going to take in the future for NAS and servers anyway, because we'll want to collect the image dimensions and meta data before users visit the app, if possible.

### Let it be

After the first use, a majority of thumbnails will have been generated and users will only have to wait for the creation of new ones. That's not so bad.
If users upload a whole new folder, then we're back to the "1st use" scenario and could apply the same strategy.

Thoughts?

@enoch85 @ezraholm50 @georgehrke @icewind1991

jancborchardt, 2016-03-31:
What’s the exact problem? The initial generation of thumbnails just takes a bunch of time? If so, I would just show an emptycontent view with spinner and h2 saying »Generating thumbnails …« below, no?

oparoz, 2016-03-31:
It's exactly the problem, but in the latest version of Gallery+, you get your first images quickly, so you know something is happening, but you just have to wait for all the images to be rendered. And every time you enter an album, it's the same scenario.
People used to ownCloud will know that it's because thumbnails are generated, but newcomers won't and will assume that Gallery is a super slow app.

jancborchardt, 2016-03-31:
Does it make sense / is it possible to generate the other thumbnails for child folders in the background?

oparoz, 2016-03-31:
It's technically possible, but I think the ratio `amount of work / benefit` will be pretty poor on these devices because we can't really have a background process running at the same time as what's needed to build and manage the UI.

* We can only load thumbnails 1 at a time, so that we can stop when the slideshow is launched or when the current view needs more thumbnails (we use lazy loading for the current album)
* We have to wait for the current view to be ready, by that time, users may already have selected another album and it could even be a different one than the one we're preloading thumbnails for

One thing I could is get rid of all the square images which are used in albums since I'm using CSS cropping now, but that gives us a 1-4 images head start per sub-album and not a proper solution

tflidd, 2016-04-06:
The RPi is a great device but it is not very powerful. Not only the gallery app can bring this device to its limits. So I would keep the gallery app as it is for the moment and provide some tips on how to optimize the system for this purpose.

There is already an occ-command to create thumbnails (https://github.com/owncloud/gallery/wiki/Create-Thumbnails). Is it possible to stop this process gracefully, or just create a fixed number of thumbnails? Then it would be easy to create a cronjob that runs every night but doesn't run over several days.