owncloud-archive/pi-image #33: 10% performance gain - Lower number of interrupts when USB is idle - dwc_otg.fiq_fix_enable=1

ezraholm50, 2016-04-12:
When this `dwc_otg.fiq_fix_enable=1` is added to cmdline.txt it should boost performance.

`Lowers the number of interrupts when USB is idle, gains about 10% performance on the ARM side`

http://www.raspberrypi.org/phpBB3/viewtopic.php?f=28&t=7866&start=75

kyrofa, 2016-04-12:
In that same thread:

>>    bitwelder wrote:
>>    Hello,
>>    I applied the patch and in my case (nothing attached to USB) interrupts went from 8000+ to 290.
>>
>>    Question: is there any known 'side effect' to this FIQ patch, or any conditions where it may not be convenient to activate it?

> That's why it's pushed out disabled with a suggestion people test it. If there's no problems we'll enable it by default. We didn't find any problems with our internal testing, although rtramon has possibly found something.

Has it been enabled by default, yet? If not, perhaps we should have the same concern.

jospoortvliet, 2016-04-13:
yeah, I suggest to let them figure it out - if they enable it by default, cool. We don't have to do this in advance... But it IS good to know there's room for improvement and future releases will make the USB thing a bit snappier!