owncloud-archive/pi-image #34: Offer an alternative image which uses a LAMP snap instead of an ownCloud snap

oparoz, 2016-04-14:
The main advantage of Snappy from my pov is that the **OS and platform/stacks** can be automatically updated and secured remotely. Some users will not like the current level of micro-management found in the ownCloud snap, while new comers and fans of set-and-forget appliances will love it.

## Current snap

### Limitations

* Most of ownCloud is on a read-only partition (all but data, apps and config)
* Impossible for users to add make any modifications to the LAMP stack
* Some configuration settings cannot be altered by users
* `.htaccess` is read-only
* Migration/rollback issues range from difficult to impossible to solve by users

Some of these items can be fixed, but it will take some extra work by community maintainers.
Overall, I'd say that ownClouders have less freedom with this solution

### Advantages

* The tighter the control, the less things can go wrong in case of ownCloud upgrades. It's much easier to replicate a failed scenario and to provide a fix compared to the hundreds of use cases ownCloud has to face today
* Some files are kept safe. Users can almost only modify non-critical files. Of course, a bad config will take ownCloud down...

## Alternative

* Working updater app, targeting the stable branch
* Working integrity checker
* Fully customizable, can even be used for development
* Ability to compile Apache and PHP modules (optional)

## Conclusion

I think the WD PiCloud can target different markets. People who want to dump their photos to their device and rely solely on the GUI and the sync clients will love the current snap, but anybody with slightly more complex requirements will get frustrated, especially when seeking help from forums or Github since most tips won't work.

I don't know who WD had in mind, so I don't know which solution is the best fit for their hardware.
It's very probable that someone will release a LAMP snap and then anybody could install ownCloud on it, but just like users can pick a generic VM as their base, the experience is much nicer when they pick the official ownCloud VM.

@jospoortvliet @enoch85 @kyrofa

bonsje, 2016-04-24:
i could not respond to your topic on the owncloud forum, so i do it her :). 
I tried to install your option 3.  

I sideload the option 3 amd64 .snap in a virutalmachine. 
But the install seems to fail (running ubuntu 16.04 amd64)

`test2@test2:~$ sudo snap install /home/test2/Downloads/owncloud_9.0.1interfasys1_amd64.snap 
[/] Make snap "owncloud" available to the system
error: cannot perform the following tasks:
- Make snap "owncloud" available to the system ([start snap.owncloud.php-fpm.service] failed with exit status 1: Job for snap.owncloud.php-fpm.service failed because the control process exited with error code. See "systemctl status snap.owncloud.php-fpm.service" and "journalctl -xe" for details.
)
`

oparoz, 2016-04-24:
Thanks for testing @bonsje . Unfortunately, these snaps don't work yet with the desktop or server versions of Ubuntu because we're using Ubuntu Core to test these images and that hasn't caught up yet. So for now you have to test using the older base images of Core linked in the post in the forums.

bonsje, 2016-04-25:
Hi Olivier,

Will the snap packages be compatible with the new Ubuntu core (16.04) soon ?
Or is this not easy possible because the Ubuntu snappy core uses a older
base  (never used ubuntu snappy core).

For input that you are asking about the Option 1/2/3 in the own
cloud forum.
I think option 3 will be the best, because we need the possibility to hack
the configs of Apache2/php and owncloud itself.

kyrofa, 2016-04-25:
> Will the snap packages be compatible with the new Ubuntu core (16.04) soon ? Or is this not easy possible because the Ubuntu snappy core uses a older base (never used ubuntu snappy core).

Oh yes-- the Snappy development team was recently focusing on the desktop. As soon as that work trickles back into Ubuntu Core that's what we'll be using.

oparoz, 2016-04-25:
Thanks for the feedback @bonsje!