owncloud-archive/pi-image #37: Deliver IDS-firewall snap

oparoz, 2016-05-06:
It should be possible to combine various software in a snap which can detect attacks and block them.

If we can find one, then it would be good to include it in the image (pre-configured for our use-case)

tflidd, 2016-06-09:
Do you have experience with such software? Do you know how much resources they need and how efficient they are (training, false positives, analysis, ...) and against which threats they are used?

Fail2ban? But with more and more ISPs using carrier-grade NAT that's perhaps not the best idea.

oparoz, 2016-06-09:
>Do you have experience with such software? 

Yes, but unfortunately with FreeBSD based software. Having said that, it's more about selecting the right tools and getting people to define the proper policies.

>Do you know how much resources they need and how efficient they are (training, false positives, analysis, ...) and against which threats they are used?

The goal is to protect the solution against scripts and to let advanced users tune the solution to their liking.
In that respect, you can have a set of basic blocking mod_security rules per example and have extra ones which only log the information.
Fail2ban is great and universal, so that would be a great feature, but as you've mentioned, it will probably only work as an IDS and the owner would still need to regularly look at logs and see if something needs to be done to block some connection attempts.

As far as performance are concerned, a firewall has a minimal impact, but mod_sec does introduce some latency...